legras/uav3i #47: Incoherent UAVDataPoint constructors?

legras, 2015-07-07:
Les différents constructeurs de UAVDataPoint traitent `alt` et `course` différemment. 

```java
    altitude  = (double) alt / 1000.;
    course    = (double) c / 10.;
```
et
```java
    altitude  = (double) alt / 100.;
    course    = (double) c;
```

phtanguy, 2015-07-07:
Je regarde en début d'après-midi, je suis pris ce matin.

phtanguy, 2015-07-07:
There is 3 different constructors for the class `UAVDataPoint` which reflect 2 different ways for obtaining the UAV position. This two ways don't use the same unit for course an altitude. (?)
 ```java
public UAVDataPoint(int utm_east, int utm_north, int utm_zone, int c, int alt, long t)
 ```
The first one is used with the microjet simulator (*3i project*) and uses the `GPS` message catched on the Ivy bus.
 ```java
@Deprecated
public UAVDataPoint(int lat, int lon, int c, int alt, long t)
public UAVDataPoint(int lat, int lon, int c, int alt, long t, double camTargetLat, double camTargetLong)
 ```
The other ones are used with the rotorcarft simulator (*berisuas project*) and use the `GPS_INT` message (and the `NAV_STATUS` message for the `course` value). These two constructors are nearly the same: the first one is now deprecated and the second one transmits the camera target position.

phtanguy, 2015-07-07:
I thought there was no problem but actually is!

phtanguy, 2015-07-07:
Fixed, see 69d79fdde5603b64117bd973fa97b7531585f9ca

closed 2015-07-07 by phtanguy